diyruz/flower #5: custom sensor with some problem

jumping2000, 2020-12-09:
I built a custom and very compact version of  your "flower" sensor  with only BM280 sensor (so it's only a temprature/hum/pressure sensor) but I have some problems with the data  coming from the sensor, I only have temperature, battery and link quality, This is the zigbee2mqtt log:

`INFO: MQTT publish: topic 'zigbee2mqtt/0x00124b001bdd56d3', payload '{"battery":100,"linkquality":54,"temperature_1":21.21,"temperature_bme":20.86,"voltage":3500}'`

So my question is: does the firmware is "fully working" with only BME sensor? 
thank you

jumping2000, 2020-12-09:
I use 2.03 firmware and related "conveerter file"

nurikk, 2020-12-20:
z2m natively supports this sensor, no external converter required

jumping2000, 2020-12-24:
> z2m natively supports this sensor, no external converter required

I removed the converter from z2m configuration but I have some strange data ending with "_1", is it normal?

{
    "battery": 52,
    "humidity": 64.48,
    "humidity_1": 47.71,
    "illuminance": 1911,
    "illuminance_1": 2615,
    "illuminance_lux": 2,
    "illuminance_lux_1": 2,
    "linkquality": 36,
    "pressure": 979.3,
    "pressure_1": 965,
    "soil_moisture": 100,
    "temperature_1": 26.61,
    "temperature_bme": 19.7,
    "voltage": 2900
}

jumping2000, 2021-01-13:
solved after 2-3 re-pairing end new version of zigbee2mqtt, Now I have the correct sensor!

closed 2021-01-13 by jumping2000