diyruz/flower #7: Connection issues

ftp27, 2021-01-13:
I've just finished my the very first device and faced with a problem with a strange behaviour when it already paired. A led continues blinking from time to time (like it in pair mode) and I don't see any updates from the device except finished interview by zigbee2mqtt and state:
```json
{
    "battery": 16,
    "soil_moisture": 100,
    "voltage": 3200
} 
```
I don't know if its hardware problem or software but PCB has the minimum amount of components, the build only for soil moisture measures. Is it compulsory to have the whole set of sensors for the firmware (v2.0.4) and what can affect to such behaviour?

mkocus, 2021-02-06:
Got similar issue. It was basically caused by issue on the PCB assembly and/or zigbee interference with BT/WiFI. 
It seems that the sensor was was not able to be configure correctly somehow.

ftp27, 2021-04-09:
So. I've soldered another board. With components
- R1, R2, R4
- C1
- D1
- LED1
- SW1
- U6
- CR2032
[image]
Only for soil humidity measure. Its still nothing but now I receive more data
```
{
    "battery": 14.5,
    "illuminance": 6,
    "illuminance_1": 16,
    "illuminance_lux": 1,
    "illuminance_lux_1": 1,
    "linkquality": 91,
    "pressure": 0,
    "soil_moisture": 0,
    "temperature_bme": 0,
    "voltage": 2900
}
```
The battery is new, but shows 14.5% Is it possible that firmware didn't adapt for CR2032 battery?

RomanDidenko, 2021-04-09:
C2 is required with CR2032

ftp27, 2021-04-09:
@RomanDidenko 
[image]

ftp27, 2021-04-09:
@RomanDidenko 
[image]
`C5` from v1 is `C2` in v2

ftp27, 2021-04-09:
After 3 hours of work the battery is dead I suppose. It stopped sending any data.

RomanDidenko, 2021-04-09:
[image: image.png]
[image: image.png]
This screenshot is taken from ready devices at https://t.me/zigberu

ftp27, 2021-04-12:
@RomanDidenko  It seems you how some problems with attachments )

RomanDidenko, 2021-04-12:
Sorry for that.
I've updated my comment but the images still not auto-expanded :(. 
As I see links are valid and one can follow them.

ftp27, 2021-04-13:
@RomanDidenko Thank you for the pictures. I'll try it

GSternik, 2021-06-09:
1. I have a problem with publishing zigbee2mqtt notification of Flowers device (Domoticz). 
When mqtt service is restarting I get: 
cze 09 10:11:02 raspberrypi npm[503]: Zigbee2MQTT:info  2021-06-09 10:11:02: 0x00124b00225734bb (0x00124b00225734bb): DIYRuZ_Flower - DIYRuZ [Flower sensor](http://modkam.ru/?p=1700) (EndDevice)
cze 09 10:11:02 raspberrypi npm[503]: Zigbee2MQTT:info  2021-06-09 10:11:02: 0x00124b0022573477 (0x00124b0022573477): DIYRuZ_Flower - DIYRuZ [Flower sensor](http://modkam.ru/?p=1700) (EndDevice)

 Zigbee2MQTT:info  2021-06-09 08:55:37: MQTT publish: topic 'zigbee2mqtt/0x00124b0022573477', payload '{"battery":100,"illuminance":534,"illuminance_lux":1,"pressure":null,"soil_moisture":100,"temperature_ds":0,"voltage":3100}'

 Zigbee2MQTT:info  2021-06-09 08:55:37: MQTT publish: topic 'zigbee2mqtt/0x00124b00225734bb', payload '{"battery":76,"illuminance":555,"illuminance_lux":1,"pressure":null,"soil_moisture":100,"voltage":3000}'
After that the sensors do not send messages for zigbee2mqtt service. 

Firstly I got message from converter: 
{"dateCode":"22/04/2021 11:13","description":"[Flower sensor](http://modkam.ru/?p=1700)","friendly_name":"0x00124b00225734bb","hardwareVersion":1,"ieeeAddr":"0x00124b00225734bb","lastSeen":1623097284351,"manufacturerID":0,"manufacturerName":"modkam.ru","model":"DIYRuZ_Flower","modelID":"DIYRuZ_Flower","networkAddress":44718,"powerSource":"Battery","softwareBuildID":"22/04/2021 11:13","type":"EndDevice","vendor":"DIYRuZ"},{"dateCode":"22/04/2021 11:13","description":"[Flower sensor](http://modkam.ru/?p=1700)","friendly_name":"0x00124b0022573477","hardwareVersion":1,"ieeeAddr":"0x00124b0022573477","lastSeen":1623224768376,"manufacturerID":0,"manufacturerName":"modkam.ru","model":"DIYRuZ_Flower","modelID":"DIYRuZ_Flower","networkAddress":56759,"powerSource":"Battery","softwareBuildID":"22/04/2021 11:13","type":"EndDevice","vendor":"DIYRuZ"}],"type":"devices"}'.

2. Now I have two DS18B20 sensors in the devices. Data pins are connected to P1.3 and Vdd to P1.1. At pin P1.1 is not any voltage for supply this sensors. My guess is that it might show up when the microprocessor is waking up. However, I don't have any sensor data.
How to resolve or debbug this problems?

bojanpotocnik, 2021-06-09:
> Now I have two DS18B20 sensors in the devices.

Try with just one connected DS18B20 sensor, this device does not support reading nor reporting values of multiple DS18B20 sensors on single 1w bus. And you are correct, sensors are powered only while being read ([source](https://github.com/diyruz/flower/blob/master/Source/zcl_app.c#L238-L254)).

It is also possible that you have [counterfeit](https://github.com/cpetrich/counterfeit_DS18B20)/dead DS18B20 (happened to me).

Also, it is better to open a new issue or using more related one. Here we are waiting for @ftp27 to report and close this issue.