diyruz/flower #9: New function or a great idea

ingo01978, 2021-01-29:
Hello, I am very excited about this project.  Now my wife got the idea to let the LED blink if the value falls below an adjustable value.  Would that be possible?  Can you set the humidity to zero by hand?  My sensors all show 20% in the dry.  LG Ingo

shajek, 2021-01-30:
Hello, i want to write about calibration. In my expirience with capacity and resistant type of probe, There is a need to use the maximum and minimum moisture calibration of the soil.  Normally is needed to bake your soil, measure minimum humidity, and water the soil to maximum  and that numbers represent as 0-100%. Is there any chance to implement min/max calibration in your great project ?

nurikk, 2021-02-01:
You're using a bad diode, try the replace it

closed 2023-03-23 by ingo01978