diyruz/flower #10: Compilation requirements

shajek, 2021-01-31:
Hello,
thank you for your work. However, can you add simple describe for IDE/enviroment where you compile FW, or what you need to compile ?

Thank you

genboo, 2021-02-01:
TI Z-Stack 3.0.2 + IAR Embedded workbench IDE (8051)

MatthijsThijssen, 2021-05-30:
Hi, Thanks for this nice project! I think I do something wrong, or maybe I am missing something...
I install the IAR Embedded Workbench environment and the Texas Instruments Z-Stack 3.0.2 but as soon as I will load the project I got a message: "Unable to create configuration "CHDTECH_DEV" using toolchain 8051" Has someone an idea? (Yes, the IAR environment and Z Stack is new for me ;-) Thanks!

bojanpotocnik, 2021-05-30:
Have you tried the steps linked by **nurikk** in issue https://github.com/diyruz/flower/issues/21?

MatthijsThijssen, 2021-06-01:
Thanks bojanpotocnik, it helps a lot. I have some compiling issues to solve (maybe because i have version 10.40 and can't find the 10.30 version)...

bojanpotocnik, 2021-06-01:
@MatthijsThijssen are you getting the following?
```
Error[e46]: Undefined external "?B" referred in bdb_Reporting ( D:\Git\diyruz\flower\CC2530DB\CHDTECH_DEV\Obj\bdb_Reporting.r51 ) 
Error[e46]: Undefined external "?IE" referred in ?ALLOC_XSTACK8 ( C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\8051\LIB\CLIB\cl-pli-blxd-1e16x01.r51 ) 
Error while running Linker 
 
Total number of errors: 2 
Total number of warnings: 11 
```

bojanpotocnik, 2021-06-01:
@MatthijsThijssen are you getting the following?
```
Error[e46]: Undefined external "?B" referred in bdb_Reporting ( ...\flower\CC2530DB\CHDTECH_DEV\Obj\bdb_Reporting.r51 ) 
Error[e46]: Undefined external "?IE" referred in ?ALLOC_XSTACK8 ( C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\8051\LIB\CLIB\cl-pli-blxd-1e16x01.r51 ) 
Error while running Linker 
 
Total number of errors: 2 
Total number of warnings: 11 
```

MatthijsThijssen, 2021-06-01:
Yes, your right! As far as i can see it is related to the current version of IAR WB 10.40 

It isn't simple to get version 10.30 and i need to figure out how to solve this so i can use 10.40 

I saw a message by TI website about this issue but didn't spend enough time to see if it could be solved...

MatthijsThijssen, 2021-06-01:
https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/999931/cc2530-cc2530-linking-issues-with-lates-iar-8051-version-10-40-1/3706070#3706070

MatthijsThijssen, 2021-06-01:
Ok, booked some succes. I changed the linker and i changed the cc2530F256.xcl file. Looks solved now. (I got a HEX file, not tested yet)

MatthijsThijssen, 2021-06-12:
Hi all, Thanks again for this nice build project! I still have some issues. I can compile the source, got a working HEX file but it differs from the original HEX file... In the original HEX file I see the LED works correct. In my compiled version the LED is very dimmed. Beside that it works but for some reason the board stopped working when the batteries are below 2.8 Volt. I think I need to do something with the compiler environment to define the correct destination (chip and/or destination environment)? Are there "thinks to know" by using the IAR development software? Thanks in advance and keep up the good work!

nurikk, 2021-07-17:
Kindly refer to this manual to setup your development environment https://zigdevwiki.github.io/Begin/IAR_install.html