diyruz/flower #17: Mention in README how to purchase these

onedr0p, 2021-04-17:
I know there's some fun to have these boards created but I don't know why it's not mentioned on the README. What is the currently way to purchase and built these devices?

bojanpotocnik, 2021-04-22:
I ordered them as described on the [author's article](http://modkam.ru/?p=1832)
[image]
That is 4.13$ delivered to my door (customs handled by JLCPCB because I selected that option and it was included in shipping price).

The only thing I needed to solder was the battery holder, which is also the easiest part to solder, or at least worth to learn a bit of soldering for the price difference.

Already happily using more than half of them :)

jamiekris, 2022-07-07:
Is there any way now to get a hold of this kit? Does this ship from Russia?

closed 2023-10-11 by onedr0p