diyruz/flower #18: E18 MS1PA2-PCB

shajek, 2021-04-19:
Hello, please, is there any FW for E18 MS1PA2-PCB ? Please, can you compile it for this module ? I need it for longer range in garden and i have it already soldered then i realize, that FW for E18 MS1-PCB is not compatibile due PA.

Thank you

nurikk, 2021-04-22:
Hi, there is no MS1PA2 build. You can build it on your own

bojanpotocnik, 2021-04-22:
I suggest you add a repeater using yours or similar module (like [this one](https://github.com/tube0013/tube_gateways/tree/main/tube_cc_router) using [RF-BM-2652P2](https://www.szrfstar.com/product/237-en.html)), but with IPEX connector and external antenna instead of PCB antenna ([any](https://www.aliexpress.com/wholesale?SearchText=ipex+antenna) will do)! Well, I don't know your case, but maybe even the simplest CC253x repeater will do.

In this case, you have the following benefits:
- You can use plenty of [Flower v3](http://modkam.ru/?p=1832) costing you <5$ per piece delivered to your door (JLCPCB assembly) which is less than cost of E18 MS1PA2-PCB alone
- Battery in sensors will last longer, because you don't need to power LNA in MS1PA2-PCB
- Repeater can have a larger battery or be connected to main power

I ordered 30 sensors and have 18 already set-up. In the house I use [this antenna](https://www.alfa.com.tw/products/ars-n19?variant=36473964986440) attached to repeater, outside I use [this one](https://www.alfa.com.tw/products/apa-m04?variant=36473963806792), directed to the garden, picking up 8 v3 sensors up to 60 m away (haven't tested further). So yeah, in my opinion it is not worth spending time and money for replacing modules - that's why repeaters are for.

shajek, 2021-04-26:
Thank you both :) Problem is, that module is already soldered and equipped with solar panel for energy demand. Programming and making electronics is favorite hobby.

bojanpotocnik, thank you for advice, specialy for v3 version. I am aware of concept of repeater, i have whole house covered, but making PA version with aditional solar charging is "hobby and fun" and maybe better experience :)

@nurikk thank you too, for advice. I dont own IAR, i try time trial version. Installed zstack, make all dependencies OK, make imports all right, but 
```
Building configuration: GenericApp - DIYRuZ_Flower 
Updating build tree... 
Performing Pre-Build Action 
OSAL_Math.s51 
Warning[410]: B is using fixed address 0xF0. Use external label ?B instead if B is using another address C:\Components\osal\mcu\cc2530\OSAL_Math.s51 108 
Warning[410]: B is using fixed address 0xF0. Use external label ?B instead if B is using another address C:\Components\osal\mcu\cc2530\OSAL_Math.s51 185 
version.c  
Linking 
Warning[w6]: Type conflict for external/entry "GetBindingTableEntry", in module APS against external/entry in module BindingTable; function return types are different; class/struct/union types have different  
sizes 
Warning[w6]: Type conflict for external/entry "bindFind", in module APS against external/entry in module BindingTable; function return types are different; class/struct/union types have different sizes 
Warning[w6]: Type conflict for external/entry "bindFindExisting", in module APS against external/entry in module BindingTable; function return types are different; class/struct/union types have different sizes 
Warning[w6]: Type conflict for external/entry "bindIsClusterIDinList", in module APS against external/entry in module BindingTable; function types differ in parameter 1; class/struct/union types have different  
sizes 
Warning[w6]: Type conflict for external/entry "bindRemoveClusterIdFromList", in module APS against external/entry in module BindingTable; function types differ in parameter 1; class/struct/union types  
have different sizes 
Warning[w6]: Type conflict for external/entry "bindRemoveEntry", in module APS against external/entry in module BindingTable; function types differ in parameter 1; class/struct/union types have different  
sizes 
Warning[w6]: Type conflict for external/entry "pbindAddEntry", in module APS against external/entry in module BindingTable; function return types are different; class/struct/union types have different sizes 
Error[e46]: Undefined external "?B" referred in bdb_Reporting ( C:\Texas Instruments\Z-Stack 3.0.2\Projects\flower\CC2530DB\DIYRuZ_Flower\Obj\bdb_Reporting.r51 ) 
Error[e46]: Undefined external "?IE" referred in ?ALLOC_XSTACK8 ( E:\8051\LIB\CLIB\cl-pli-blxd-1e16x01.r51 ) 
Error while running Linker 
 
Total number of errors: 2 
Total number of warnings: 9 

```

because of this, i kindly ask, if you can make hex for PA2 (i know that, for CO2 meter is available)

thank you both

nurikk, 2021-05-04:
I don’t have time for PA build implementation.
https://zigdevwiki.github.io/Begin/IAR_install/

@shajek you should use IAR v 10.30 or 10.40

closed 2021-05-04 by nurikk