diyruz/flower #19: Query U1 U2 U3

aidbish, 2021-04-23:
In the V2 picture
https://github.com/diyruz/flower/blob/master/images/DIYRuZ_Flower_V2-5.png

Are U1 U2 U3 no longer required as listed in the BOM, but in picture not installed?

grego1981, 2021-04-23:
I'd like to know if there is a flash defentive flash guide!

grego1981, 2021-04-23:
I have used the guide in this URL -->  https://zigbee.blakadder.com/Ebyte_E18-MS1-PCB.html
using the ccdebuger method --> https://ptvo.info/how-to-select-and-flash-cc2530-144/

The debuger flashes the image with error in verification (CC2530 - ID1258: Error in HEX file verification) the module does not seem to work... Is there any test file  to check?? Anyhting to test prior to flashing?

linuxrrze, 2021-04-24:
AFAIK U1 and U2 are optional sensors. If equipped their values are reported, if not (default) their values are "0".

grego1981, 2021-04-24:
OK had a breakthrough, I have managed to flash and read the chip by connecting the cc debuger to the board, ommiting the power link. I connected the battery as a power input for the chip. Now it flashes and passes the verification.

The problem of connection remains, I cannot make it connect to the gateway, I tried holding the button more than 10secs to reset to no avail.... How do youmake it connect!!??

grego1981, 2021-04-28:
The problem was the button, it was defective, I have changed it and it works when long pressing it both the VD1 and LED1 light up every 3 secs (while holding) but HA zigbee2mqtt does not find the device... Is there any procedure tojoin? Anyway to debug the behaviour?

bojanpotocnik, 2021-04-28:
@grego1981 are you sure about VD1 (PTSMD3528) also lighting up? :raised_eyebrow: 

Otherwise, procedure [described here](https://github.com/diyruz/flower#how-to-join) + clicking _Permit join (All)_ in Zigbee2mqtt works for me.

grego1981, 2021-04-28:
Yes, It was to curious to me as well.... When I press and hold it flasshes 1 time and then three faster, then it does not flash at all. I'll try to upload a video of it..

grego1981, 2021-04-28:
https://drive.google.com/file/d/14oaWWbRJhI-b1aqj9ErwmWVT1dMnpDkP/view?usp=sharing

bojanpotocnik, 2021-04-28:
I tried my V3 sensors with power supply set to 3.00V/1.0mA and 3.30V/3.0mA: while LED1 shines bright white in both cases, phototransistor VD1 does not light up. Weird that yours actually shines.

Anyway, I noticed that there is IPEX/U.FL connector on your module. Are you using [E18-MS1-IPX](https://www.ebyte.com/en/product-view-news.html?id=124)? In that case you shall attach the [antenna](https://www.aliexpress.com/wholesale?SearchText=antenna+ipex), otherwise RF part cannot work.

grego1981, 2021-04-28:
Ah this does makes sense..
I bought the modules bellow and have the V2 boards.
https://www.ebay.com/itm/192164529297

grego1981, 2021-04-28:
> I tried my V3 sensors with power supply set to 3.00V/1.0mA and 3.30V/3.0mA: while LED1 shines bright white in both cases, phototransistor VD1 does not light up. Weird that yours actually shines. [...]

You were correct, I found and installed a 2.4 sma antenna and managed to join it. Now Have to find out why no sensor works... 
Thank you for your help!

grego1981, 2021-05-04:
So the sensor got discovered and at some point the battery was detected but low (about 12%) I even saw some data from the sensors briefly but after some time, all data was N/A and only the signal strength is displayed... I deleted and rejoined to no avail, I changed the battery as well. 
I use the latest released hex file.

bojanpotocnik, 2021-05-04:
Can you please configure zigbee2mqtt as such
```
advanced:
  log_level: debug
zigbee_herdsman_debug: true
```
and then provide log output after you re-join the sensor? Force remove it first, rejoin (as described in README.md in this repo) and then press the button after a while to update the state.

robotm-team, 2021-06-07:
> AFAIK U1 and U2 are optional sensors. If equipped their values are reported, if not (default) their values are "0".

I will ask because I don't know if I understand this correctly:
Will a plate made as in the photo DIYRuZ_Flower_V2-5.png, that is without U1, U2 and U3 will it only report soil moisture and insolation?
Installing U2 will add ambient temperature report?
Installing U1 or U3 will add air pressure, humidity and temperature but only ambient?
Does it make sense to install U2 if you install U1 or U3?

bojanpotocnik, 2021-06-08:
> Does it make sense to install U2 if you install U1 or U3?

I think it is meant that you add this (left one) as U2
[image]
and stick it in the soil - then you have soil humidity+temperature + air humidity+pressure+temperature, which makes sense again.

> nothing appears in the debug log

Have you enabled all debug as I suggested above? And LED flickers like it is sending something, but totally nothing in the log (joining new devices permitted)? In that case I am also run out of ideas... I would try with another sensor first, to test HA/Z2M setup.

robotm-team, 2021-06-20:
Hi. One more question.
From what I've read VD1 (3528-PTSM D3528) which I can't find anywhere except LCSC and supplying only this one element from there is completely unprofitable, can be replaced with any photoresistor ( https://modkam.ru/?p=1700#comment-6226 ) like the one in the link in the quoted comment. However, there is a statement which I do not understand. I don't know Russian and I'm using online translator and it's not that helpful.
Question: What should be done with the resistor (I think it's R3) in the circuit Jager writes about?
"Только резистор в цепи подберите." - What to do with it :-) ? omit it (make a jumper or solder 0Ω) ? Give more than 10KΩ ? Sorry, but the translator suggests all these possibilities ...

bojanpotocnik, 2021-06-20:
When measuring illuminance, ADC peripheral [is set](https://github.com/diyruz/flower/blob/master/Source/zcl_app.c#L288) to use AVDD as a reference voltage, which is connected to VDD, meaning that voltages up to 3.3 V can be read on ADC pins. However, the circuit will work down to 2V VDD when battery empties, so you would want to limit your measured voltage to max. 2V to prevent saturating ADC input.

Checking [3528-PTSM D3528 datasheet](https://datasheet.lcsc.com/lcsc/1912271606_JCHL-Shenzhen-Jing-Chuang-He-Li-Tech-PTSMD3528_C242256.pdf) page 3/5

[image]
[image]

shows you that at 100 lux the current is 59 uA, resulting in ADC input voltage U = I*R = 59 uA * 10 k = 0.59 V.

So if you want to have the same values, you should measure photoresistor's resistance at 100 lux and then adjust the R3 so that the voltage divider output will be 0.59 V. But then you have another problem - LDRs are usually not linear
[image]
so values won't be the same anyway, so don't bother too much. Just measure the resistance at maximum brightness and adjust R3 so that the resulting value does not go over 2V.

As example, if you take LDR info from [this page](https://www.elprocus.com/ldr-light-dependent-resistor-circuit-and-working/)
> Typical resistance at 100lux is 0.7kΩ

and put this in voltage divider [calculator](https://ohmslawcalculator.com/voltage-divider-calculator) (Vs = 3V, R1 = 700 ohm, Vout = 2 V), you get that R2 (R3) shall be 1400 ohm.

robotm-team, 2021-06-20:
Thank you so much for your help and commitment.
So, if I understand correctly this circuit is fitted to this particular type and model. Well, too bad, I will have to do without it.Unless there is some other, more easily available model of photoresistor which can simply be soldered in and will match the parameters... From the calculator it looks like it should be 5kΩ at 100Lx to match R3 (10kΩ) on the board.
Maybe someday, additionally...
Unless you have a ready solution :-) ?

Translated with www.DeepL.com/Translator (free version)

MatthijsThijssen, 2021-06-20:
This one works well:

https://nl.rs-online.com/web/p/phototransistors/7103853/

Or this one:

https://nl.rs-online.com/web/p/photodiodes/1652707/

Met vriendelijke groet / Kind Regards,
 
Matthijs Thijssen
The Netherlands 
 

(Verstuurd vanaf mijn iPhone)

bojanpotocnik, 2021-06-21:
Also check [this comment](https://modkam.ru/?p=1700#comment-5586) - you can add [I2C light sensor](https://ae01.alicdn.com/kf/HTB1HVxiQpXXXXcWaXXXq6xXFXXXx.jpg) instead of or in addition to BME280. This would result in calibrated light measurements, but also in increased cost and need of modifying the firmware.