diyruz/flower #20: Feature Request: OTA Updates

linuxrrze, 2021-04-25:
Not a bug, just a feature request: OTA Updates

I'd love to see OTA updates for firmware releases.
Are there any plans to realize that?

So far I've only seen that on a few IKEA devices, however it'd be great to have this feature for more devices (especially if there's active development going on as in this project).

There's some documentation about how OTA can be done (I hope the mentioned Z-Stack is the same as the one used here):

https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/158/6888.Z_2D00_Stack-OTA-Upgrade-User_2500_27s-Guide.pdf

PS: Thanks for sharing this great project!

nurikk, 2021-04-26:
Hi, unfortunately there’s not enough flash size for ota

chrisrock1984, 2021-04-27:
would it still be possible to do OTA if the firmware only contains the code of moisture sensing?

nurikk, 2021-04-27:
I'm not sure about this, unfortunately I don't have time for this experiments and implementation.
If you want to try implement OTA by yourself kindly refer to TI official reference guide. All contributions are always welcome

[Z-Stack OTA Upgrade User's Guide.pdf](https://github.com/diyruz/flower/files/6382002/Z-Stack.OTA.Upgrade.User.s.Guide.pdf)

closed 2021-05-04 by nurikk

Hedda, 2021-11-29:
FYI, OTA would be possible if this Flower project was redesigned to newer Texas Instruments module like CC2652, see -> https://github.com/diyruz/flower/issues/43