diyruz/flower #21: How to compile?

chrisrock1984, 2021-04-27:
Hi,
this is a great project. I already have 7 sensors in use and 10 are going to be build.
I was wondering how to set up a development environment?
Furthermore, I tried opening the source code with VSCode as there is a .vscode folder in it but failed.
What do I need in addition?

nurikk, 2021-04-28:
Here is very detailed how to manual https://zigdevwiki.github.io/Begin/IAR_install/

chrisrock1984, 2021-04-28:
Perfect!
Thank you very much!!

closed 2021-04-28 by chrisrock1984

MatthijsThijssen, 2021-05-31:
Thanks Chrisrock, i am sure this makes sense! I will give it a new try ;-)

DirkTas67, 2021-11-19:
> Here is very detailed how to manual https://zigdevwiki.github.io/Begin/IAR_install/

Is there a possibility that some compiles this firmware for a standard V3 version of the sensor (the one made by Jager) . The manual only handles a windows installation but I don't use windows or had windows so there is a problem.

Regards
Dirk