diyruz/flower #22: Cut down the cost of the PCB with panelization

elysion, 2021-04-27:
PCB fabrication services charge extra for PCBs that do not fit inside 10x10cm. As the width of the v2 design seems to be 128mm, additional charges are applied. Also as the height of the board height is 24mm, you could fit 4 of these inside a single design, further cutting the costs. I realise that the cost of the PCB is one of the cheapest parts of the design, but think that small tweaks would have a good ROI.

Would it be possible for you to make these changes or share the PCB designs in order for someone else to make them?

bojanpotocnik, 2021-04-27:
@elysion check [this comment](https://github.com/diyruz/flower/issues/17#issuecomment-825236065) - v3 is less than $5 fully assembled (except battery holder) delivered to your door - it hardly gets any cheaper. Maybe try using other fabrication service?

Otherwise this is still valid request, maybe someone is willing to do it anyway. Note, however, that sensing element should be of certain minimum size to be reliable, and you want RF part to be as away from the soil as possible. Interesting design would be separating the logic/RF/upper and sensing/lower part, each having pads to connect them with a e.g. ribbon cable. That would enable using different sizes of sensing elements (smaller or larger pots) or mounting RF part on the outside of the pod (not sticking up).

nurikk, 2021-05-04:
Kindly do panelisation and submit gerbers in 0R

closed 2021-05-04 by nurikk

bojanpotocnik, 2021-05-04:
@nurikk is it possible to publish original PCB design files or make EasyEDA project public? It is much harder to reverse engineer the gerber files and not much sense in reinventing the wheel. If the original project files would be available, I am sure someone (even me when I find some spare time) would be happy to do such improvements.

bojanpotocnik, 2021-06-07:
@nurikk do you have any information on my question above, or shall I ask Jager on Telegram?

nurikk, 2021-07-22:
Jager never shares sources, forget about it