diyruz/flower #26: is it possible to use "Capacitive soil moisture sensor v1.2" with e18-ms1-pcb and this code?

errdos, 2021-05-17:
more of a question than an issue.
i was wondering if its possible (and how) to use the firmware with e18-ms1-pcb and a "Capacitive soil moisture sensor" with 3 pin 
connectors.

looking over the board it seems very very similar to the sensor that i already have (Capacitive soil moisture sensor v1.2).

bojanpotocnik, 2021-05-18:
You are correct, the shape and operation principle is basically the same as for other capacitive soil moisture sensors. As written in the [README.md](https://github.com/diyruz/flower#user-content-version-1):
> The circuit for measuring soil moisture is typical for this kind of device and, in principle, repeats the Chinese sensor, with the only difference that Jager increased the frequency of the generator to 2 MHz.

If you compare [V1 schematic](https://github.com/diyruz/flower/blob/master/images/DIYRuZ_Flower_4.png) with you v1.2 sensor, you will see that the upper left part is already present on your sensor. You can try to connect analog output of your sensor to pin P0.4.

eitanV81, 2021-05-18:
first of all thanks for the info and help!

so if i read the schema correctly i can (as you mentioned) connect p0.4 to analog output and p1.1 to VDD? (guessing from the schema of version1)
i assume that the new version P1.4 is used to create up to 3 MHZ signal instead of the chip (which already exists in my sensor)?
so the questions i am left with are:
1. is the analog input of the board support up to 3.3v (i assume it does)
2. does the VDD power still works (and enough ) using P1.1
3. or should i simply jump over the whole chip of my sensor and try to implement what is done in V2.. but i find it a bit complicated to understand what does exists in my sensor and what doesn't.

do you think it's worth pursuing?
i will also try to read on Zigbee Configurable Firmware but it seems that it doesnt support sleep :( so i wont be able to use it on battery.

errdos, 2021-05-18:
first of all thanks for the info and help!

so if i read the schema correctly i can (as you mentioned) connect p0.4 to analog output and p1.1 to VDD? (guessing from the schema of version1)
i assume that the new version P1.4 is used to create up to 3 MHZ signal instead of the chip (which already exists in my sensor)?
so the questions i am left with are:
1. is the analog input of the board support up to 3.3v (i assume it does)
2. does the VDD power still works (and enough ) using P1.1
3. or should i simply jump over the whole chip of my sensor and try to implement what is done in V2.. but i find it a bit complicated to understand what does exists in my sensor and what doesn't.

do you think it's worth pursuing?
i will also try to read on Zigbee Configurable Firmware but it seems that it doesnt support sleep :( so i wont be able to use it on battery.

bojanpotocnik, 2021-05-19:
> i can (as you mentioned) connect p0.4 to analog output and p1.1 to VDD? (guessing from the schema of version1)

Yes, that should work. P1.1 is used instead of V3.3 to decrease power consumption - so the sensor is powered (3.3V from P1.1) only for short time.

[CC2530 datasheet](https://www.ti.com/lit/ds/symlink/cc2530.pdf) pages 11, 18.

> 1. is the analog input of the board support up to 3.3v (i assume it does)

Yes, input voltage is 0-VDD V, where VDD is voltage on AVDD5 pin.

> 2. does the VDD power still works (and enough ) using P1.1

```plain
| P1_1 | 9 | Digital I/O | Port 1.1 – 20-mA drive capability
```
Yes, P1.1 is one of the 2 IO pins capable of 20 mA. Your sensor shall consume much less, but you can measure to be sure.

> 3. or should i simply jump over the whole chip of my sensor and try to implement what is done in V2.. but i find it a bit complicated to understand what does exists in my sensor and what doesn't.

That would work. You can compare V1 schematic with your sensor, then V1 with V2 to see changes. Power consumption would be lower for sure, because you don't need to power 555 on your sensor. You would connect P1.4 to where pin 3 of 555 chip goes on your sensor, but first make sure that non-powered OUT pin of existing 555 does not consume anything.