diyruz/flower #27: Проблема с датчиком DS18B20

gumileeev, 2021-06-12:
Для v3 мною были закуплены датчики DS18B20 по этой ссылке: https://aliexpress.ru/item/1783091649.html
После их установки - я получаю только 2 значения с датчика температуры: 0 и 25 градусов, значения рандомные, ни от чего не зависят. Я предполагаю, что датчики могли быть неоригинальные. Прошивку использую 2.08. Пробовал 2.05, но ситуация не меняется. В отзывах пишут о том, что должно быть какое то "паразитное" питание. Подскажите, есть ли возможность исправить эту проблему?

bojanpotocnik, 2021-06-14:
Similar happened to me - one batch of cheap sensors (20 pcs) from AliExpress always show exactly 25.0°C. You have [counterfeit DS18B20](https://github.com/cpetrich/counterfeit_DS18B20) sensors... Read reviews before buying, hope for some luck (I bought many OK DS18B20 from eBay and AliExpress in the past), or buy from legit suppliers (LCSC, Farnell, DigiKey, RS, Mouser, local electronic components stores, ...).

gumileeev, 2021-06-17:
> Similar happened to me - one batch of cheap sensors (20 pcs) from AliExpress always show exactly 25.0°C. You have [counterfeit DS18B20](https://github.com/cpetrich/counterfeit_DS18B20) sensors... [...]

Спешу сообщить, что уважаемый Jager поделился со мной другой версией прошивки 2.0.8, к ней прилагался скрин, возможно по нему вы сможете определить, какая именно это версия файла. С ней все мои датчики исправно заработали и больше не показывают 0 и 25. Вдруг вы сочтете внести такое исправление в основной релиз :)  Спасибо!
[DIYRuZ_Flower.zip](https://github.com/diyruz/flower/files/6671950/DIYRuZ_Flower.zip)
[image]

bojanpotocnik, 2021-06-18:
Nice! :open_mouth:  Maybe my "dead" sensors will also start working.

> Suddenly you will consider making such a fix in the main release :)

I am not sure if I understand this (can't read cyrillic) properly... Did you mean to say that @nurikk will push and merge the commit/fix soon and make version 2.0.9? I personally cannot do it, I don't know what the fix is.

gumileeev, 2021-06-25:
> Nice! 😮 Maybe my "dead" sensors will also start working.

У вас получилось? Сенсор заработал?

> Did you mean to say that @nurikk will push and merge the commit/fix soon and make version 2.0.9?

Да, я бы хотел, что бы внесли официальное исправление. Это не в моей компетенции.

gumileeev, 2021-07-05:
Сенсор температуры отлично отправляет данные на этой прошивке. Однако за 4 дня полностью расходует заряд 2хААА батареек. В следствии чего избавился от этих не оригинальных датчиков температуры и накатил оригинальную прошивку. Запрос закрываю.

closed 2021-07-05 by gumileeev