diyruz/flower #28: Help writing groovy device driver

AutomateMyLifee, 2021-06-24:
I'm trying to get this sensor working with the Smartthings/Hubitat platform. It paired easily to the hub, but I had to create my own driver. So far I have the sensors accurately reporting battery and values from BME280 (temp, humid, and pressure), but I can't get the sensor to send the correct values for the soil moisture. These are the commands I sent to configure the device so it sends data.
[image]
But the soil cluster (0408) is sending the attribute 0201 instead of 0000. I had it reporting the soil moisture at one point, but I must  be sending the wrong config because now it stopped sending the correct value. I checked all source code best I could, but I don't really have experience with ZigBee firmware. What am I doing wrong with the config command?
[image]

bojanpotocnik, 2021-06-28:
I think the problem may actually be in the SmartThings/Hubitat.

All working clusters IDs [are defines from `zcl.h`](https://github.com/diyruz/flower/blob/master/Source/zcl_app.h#L41-L47), that is standard [ZCL General Clusters](http://software-dl.ti.com/simplelink/esd/simplelink_cc13x2_26x2_sdk/4.20.00.35/exports/docs/zigbee/doxygen/zigbee/html/group___z_c_l___g_e_n_e_r_a_l___c_l_u_s_t_e_r___i_d_s.html) or [ZCL Measurement & Sensing Clusters](http://software-dl.ti.com/simplelink/esd/simplelink_cc13x2_26x2_sdk/4.20.00.35/exports/docs/zigbee/doxygen/zigbee/html/group___z_c_l___m_s___c_l_u_s_t_e_r___i_d_s.html#details) (there is also a reference to the same ZCL standard on the [SmartThings page](https://docs.smartthings.com/en/latest/device-type-developers-guide/zigbee-primer.html#useful-zigbee-references)).

`SOIL_HUMIDITY` cluster is not part of the [ZCL v7 specification](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf), that is why its ID is the only one [manually typed](https://github.com/diyruz/flower/blob/master/Source/zcl_app.h#L45) in source, instead of using define from `zcl.h`. Consequently it is also not present within [SmartThings Capabilities](https://docs.smartthings.com/en/latest/capabilities-reference.html#capabilities-taxonomy).

[ZigBee Cluster Library Specification](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf):
- For BME, _**0x0405** Relative Humidity Measurement - **0x0000** MeasuredValue_ (**4.7.2.2.1**) is used
- For soil moisture, nonexistent cluster _**0x0408** Soil Humidity Measurement? - Relative Humidity in % - **0x0200-0xFFFE** Vendor defined_ (**3.14.11.19.1.2**) is used

So the behaviour is correct and per-specs: because Soil Humidity is not yet supported in the Zigbee standard, author chose vendor defined Relative Humidity type 0x0201 out of range 0x0200-0xFFFE for vendor defined types.

bojanpotocnik, 2021-07-24:
@AutomateMyLifee try with the new release [2.1.0](https://github.com/diyruz/flower/releases/tag/2.1.0), if there will be any difference.

AutomateMyLifee, 2021-07-26:
I haven't found any more success. The new firmware give the same behavior, except there is some warning about the endpoint 2. So you are saying that the value from cluster 0x0408 and attribute 0x0201 should give soil humidity? I tried putting the sensor in water and in air and this value stays basically the same. Isn't it in the [source code](https://github.com/diyruz/flower/blob/e4cb8726fa5d75605b4298160d12e1dfa0b26df1/Source/zcl_app.c#L273) that the attribute should MeasuredValue, which is 0x0000?