diyruz/flower #29: Source code issue

MatthijsThijssen, 2021-07-03:
Hi all,

First of all: I would like to thank all of you working on this nice project!

To implement it in my environment i would like to make some changes. Before i can do this i try to compile the actual source version but i have some issues...

First of all: I build the developemt environment exactly as described. So far so good. I can compile the source code but for some reasons i have the following issues:

- The output Flower_CHDTECH.HEX file is 536KB in relation to the posted DIYRuZ_Flower.HEX, which is 455 KB in size...
- The consumed power of the generated Flower_CHDTECH.HEX is much higher which makes that battery period is much shorter.
- The LED is nearly not visible, this is in the org DIYRuZ_Flower perfect, so it is an  firmware issue...

Maybe there are more different things, but this is what i see after compiling the posted source (without changing any code).

Are there "Things to know and define" in the development environment before compile?
Is the source outdated in relation to the available HEX file?

Please let me know, i like the porject but i need to have it current before i can start to implement it and work on it...

Thanks in advance!
Matthijs

bojanpotocnik, 2021-07-16:
> Things changed, you need IAR version 10.30...

@MatthijsThijssen, I suggest you recover your original comment and add this as and EDIT: below. 

I think we will resolve this eventually. Personally I wouldn't be happy with myself providing "just use older version of the tool" as a requirement or solution. Things do change, but in case of compilers usually for the better. I have IAR 10.40.1 installed, have compiled this repo successfully, but haven't had a time to test and debug it yet. It seems like also @nurikk cannot dedicate much time to this project, but eventually someone will find a fix for newer version and this issue could be properly closed. Maybe you can change the title for something more related, e.g. "Compiling using IAR version higher than 10.30".

Relates to #10.

nurikk, 2021-07-16:
Nothing has been changed in zstack 3.0.2 since 2007... So sticking to old version or IAR is a good idea.

Here is download link
http://netstorage.iar.com/SuppDB/Protected/PRODUPD/013455/EW8051-10301-Autorun.exe

TPertenava, 2021-07-16:
Hi,
Trying to get hex compiled from source to move on to fixing clusters issue for DeCONZ integration.
But unfortunately output hex I am getting after compilation seems to be broken. The same device works fine when I use precompiled hex, but my own version result in solid LED after reflashing and blinking LED after the button press. No pairing (
There were some warnings when compiling, anyone has a clue what could be the issue?
`Warning[Pe047]: incompatible redefinition of macro "INP" (declared at line 10 of "C:\ C:\Texas Instruments\Z-Stack 3.0.2\Components\osal\include\comdef.h 63 
Texas Instruments\Z-Stack 3.0.2\Projects\zstack\HomeAutomation\Flower\zstack-lib\ 
utils.h") `

`Warning[Pe128]: loop is not reachable C:\Texas Instruments\Z-Stack 3.0.2\Components\hal\common\hal_assert.c 245 `

`Warning[w6]: Type conflict for external/entry "GetBindingTableEntry", in module APS  
against external/entry in module BindingTable; function return types are different;  
class/struct/union types have different sizes` 

`Warning[w6]: Type conflict for external/entry "bindFind", in module APS against  
external/entry in module BindingTable; function return types are different;  
class/struct/union types have different sizes `

`Warning[w6]: Type conflict for external/entry "bindFindExisting", in module APS against  
external/entry in module BindingTable; function return types are different;  
class/struct/union types have different sizes `

`Warning[w6]: Type conflict for external/entry "bindIsClusterIDinList", in module APS  
against external/entry in module BindingTable; function types differ in parameter 1;  
class/struct/union types have different sizes `

`Warning[w6]: Type conflict for external/entry "bindRemoveClusterIdFromList", in  
module APS against external/entry in module BindingTable; function types differ in  
parameter 1; class/struct/union types have different sizes `

`Warning[w6]: Type conflict for external/entry "bindRemoveEntry", in module APS  
against external/entry in module BindingTable; function types differ in parameter 1;  
class/struct/union types have different sizes `

`Warning[w6]: Type conflict for external/entry "pbindAddEntry", in module APS against  
external/entry in module BindingTable; function return types are different;  
class/struct/union types have different sizes `

nurikk, 2021-07-17:
Hello, this warnings absolutely harmless. Kindly refer to this manual to setup your development environment https://zigdevwiki.github.io/Begin/IAR_install.html

bojanpotocnik, 2021-07-17:
@nurikk is this page lately 404 only for me?
[image]

nurikk, 2021-07-17:
My bad https://zigdevwiki.github.io/Begin/IAR_install/

TPertenava, 2021-07-17:
> Hello, this warnings absolutely harmless. Kindly refer to this manual to setup your development environment https://zigdevwiki.github.io/Begin/IAR_install.html

OK, that means it should work but it doesn't, just the same solid LED after flashing to the device (or rebooting it) and blinking LED when pressing the button. I've compared precompiled hex files from this repo with file compiled by me and it is a lot different (it has the same size but contents have just a few similar raws, all in "red" when trying to diff it... 
Reinstalled IAR environment 10.30 from scratch and also cloned repo once again with no avail.
Also tried to glance into Project options, seems like it has everything preconfigured as expected (some custom options for Compiler, Linker, etc...)
Any idea what could be my issue?

nurikk, 2021-07-18:
Have you applied SDK patch? (Mentioned on wiki page)

Apply SDK patch
After installing the IAR, you should apply the patch to correctly compile the sources.

You can find the patch [here](https://github.com/diyruz/AirSense/blob/master/0001-Fixes.patch)

How-to install:

Save 0001-Fixes.patch to your Z-stack working directory (C:\Z-Stack 3.0.2)
Open console at Z-stack working directory
Run git apply 0001-Fixes.patch

TPertenava, 2021-07-18:
> Have you applied SDK patch? (Mentioned on wiki page)
>
> Apply SDK patch
> After installing the IAR, you should apply the patch to correctly compile the sources.
>
> You can find the patch [...]

I did, even verified manually that it was actually made changes to the z-stack files ((

nurikk, 2021-07-18:
Can you confirm that you’re using correct build profile(DIYRuZ flower)?

nurikk, 2021-07-18:
Complete rebuild all log might help me

TPertenava, 2021-07-18:
> Complete rebuild all log might help me

Hi @nurikk, really appreciate you help. I've uploaded a full build log here
https://github.com/TPertenava/flower/blob/master/firmwares/buildlog_Flower.log

nurikk, 2021-07-19:
> > Complete rebuild all log might help me
> 
> Hi @nurikk, really appreciate you help. I've uploaded a full build log here
> https://github.com/TPertenava/flower/blob/master/firmwares/buildlog_Flower.log

You're using incorrect workspace, you need to use DiYRuZ_Flower

[image]

TPertenava, 2021-07-19:
> You're using incorrect workspace, you need to use DiYRuZ_Flower

Wow, it worked, thanks a bunch!!

luchuanjie, 2021-07-20:
Error executing file

error: corrupt patch at line 46

Armin-Ka, 2022-01-15:
I have the same issue as TPertenava with several Warnings. 

Warning[Pe047]: incompatible redefinition of macro "INP" (declared at line 10 of "D:\ti\Z-Stack 3.0.2\Projects\zstack\HomeAutomation\flower-2.1.0\zstack-lib\utils.h") D:\ti\Z-Stack 3.0.2\Components\osal\include\comdef.h 63 

Warning[Pe128]: loop is not reachable D:\ti\Z-Stack 3.0.2\Components\hal\common\hal_assert.c 245 

Warning[w6]: Type conflict for external/entry "GetBindingTableEntry", in module APS against external/entry in module BindingTable; function return types are different; class/struct/union types have different sizes 

Warning[w6]: Type conflict for external/entry "bindFind", in module APS against external/entry in module BindingTable; function return types are different; class/struct/union types have different sizes 

Warning[w6]: Type conflict for external/entry "bindFindExisting", in module APS against external/entry in module BindingTable; function return types are different; class/struct/union types have different sizes 

Warning[w6]: Type conflict for external/entry "bindIsClusterIDinList", in module APS against external/entry in module BindingTable; function types differ in parameter 1; class/struct/union types have different sizes 

Warning[w6]: Type conflict for external/entry "bindRemoveClusterIdFromList", in module APS against external/entry in module BindingTable; function types differ in parameter 1; class/struct/union types have different sizes 

Warning[w6]: Type conflict for external/entry "bindRemoveEntry", in module APS against external/entry in module BindingTable; function types differ in parameter 1; class/struct/union types have different sizes 

Warning[w6]: Type conflict for external/entry "pbindAddEntry", in module APS against external/entry in module BindingTable; function return types are different; class/struct/union types have different sizes 

The Fix is applied an modifictions in the two Files are Done. I use the right Workspace.