diyruz/flower #33: Flashing new firmware

habitoti, 2021-07-25:
Now that the firmware was fixed so that the sensor can potentially work with a Cobee/DeConz setup: how would I flash it to the device? Seems not to support OTAUs…

nurikk, 2021-07-26:
You can follow this manual

https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html

habitoti, 2021-07-26:
Ok, so much for the programming equipment. Ordered, thanks!
Now I am not a total dummy wrt HW and SW, but probably at least half of it 😬

Do you also have some instructions at hand where to connect the debugger cables to? The sensor has no connector plugs whatsoever (on the other side there's a piggy backed battery holder), so I probably need to solder at least temporarily some cables to VCC/RST/P2.2/P2.1/GND, right? But tbh, I have no clue...

[image]

guny74, 2021-07-30:
I also have a CCDebugger now, but need to know how to wriring to the sensor.

habitoti, 2021-07-30:
Found this site: https://ptvo.info/how-to-select-and-flash-cc2530-144/

Received my CCDebugger today, so will try this weekend...

guny74, 2021-07-30:
Ok, I found something, but not my language https://kvshome.ru/?p=2706  I will try via the pictures...

habitoti, 2021-07-30:
See link above, it shows exactly how how to wire it...

guny74, 2021-07-30:
Flashing successful 

Thx