diyruz/flower #35: Does not work with ZHA (Zigbee Home Assistant)

qneverless, 2021-08-03:
Hi,

I have tested the device and it works correctly with Zigbee2MQTT, but not ZHA.
The device can be added to ZHA:
[image]
Unfortunately, no sensor data is recognized or displayed by Home Assistant.
Other sensors and lights are working fine, so I think there might be some quirk or non-standard behavior that needs patching.

Please help with finding the root cause of this issue.
I have an embedded programming background, but I am not sufficiently familiar with Zigbee to quickly jump into the problem.

MatthijsThijssen, 2021-08-03:
Dear sender,

I will be out of the office starting 03 August through 16 August returning. During this period i have limited access to my email. I will respond to your emails as soon as possible upon my return.
You can also send your message to ***@***.*** so one of my colleague can follow up.
Thank you for your message

nurikk, 2021-08-04:
Hello, try to use latest version 2.1.0

visstro, 2021-08-04:
I got the assembled and "ready to use" device. I guess that I will need CC debugger for flashing? Or is OTA possible?

nurikk, 2021-08-05:
Yes, you need a cc debugger. You can find flashing instructions and some support here https://github.com/diyruz/flower/issues/33

pooyashahidi, 2021-08-09:
running the latest version (2.1.0)
I think I have the same problem. I am able to add the device to ZHA but it only shows the following entities:
[image]

The illuminance value looks wrong battery started at 128% and is now 32% which also does not make sense.

If I'm not mistaken the humidity, pressure and temperature entities should come from the bme280 sensor that I do not need and did not install and another temperature is for the other optional sensor. so am I correct to assume that the whole soil sensing thing is missing?

Any help is highly appreciated :)

madman2143, 2021-08-19:
I have the same issue. The soil moisture is not reported when using ZHA.