diyruz/flower #38: Zigbee 3.0 firmware (on CC2530) for EFEKTA Zigbee Plant Watering Sensor?

Gamester17, 2021-09-17:
Please consider building and releasing a Z-Stack 3.0.x (Zigbee 3.0) firmware for CC2530 based EFEKTA Zigbee Plant Watering Sensor.

FYI, kirovilya from DIYRUZ community have backported Z-Stack 3.0 (with Zigbee 3.0) support to CC2530 as a proof-of-concept:

* https://github.com/diyruz/diyruz_rt
  * https://github.com/diyruz/diyruz_rt/releases
    * https://zigbee.blakadder.com/DIYRuZ_RT.html

kirovilya tested Z-Stack 3.0.x firmware with DS18B20 on Sonoff BASICZBR3 and also made config to be a Zigbee 3.0 router device:

* https://habr.com/ru/company/iobroker/blog/495926/ (written in Russian so used Google Translate)

For a battery-operated device you would of course want to use a Zigbee end device config and not a Zigbee router config.

More information on Z-Stack 3.0 for CC253x (CC2530 and CC2531) here:

https://www.bestov.io/blog/zigbee-3-part-1-getting-started-with-cc2530

https://www.cnx-software.com/2020/05/25/zigbee-firmware-news-ti-z-stack-3-0-zigbee-for-cc2530-ptvo-zigbee-fw-configuration-tool/

https://github.com/Koenkk/zigbee2mqtt/issues/1445

Gamester17, 2021-09-17:
By the way, it looks like Tech4You / TrimVa ZeeFlora DIY Soils Moisture ZigBee sensor is also based DIYRuZ Flower and provides/provided a Zigbee 3.0 firmware based on Z-stack 3.0.2 for Ebyte E18-MS1-PCB module based on CC2530

https://zigbee.blakadder.com/Zeeflora.html

https://github.com/TrimVa/ZeeFlora

https://github.com/TrimVa/ZeeFlora/commits/main

Perhaps @TrimVa (S. Girard & P. Schuler) would be willing to confirm or even share their Z-stack 3.0 firmware code?

PS: I also posted this request/question to smartboxchannel / EFEKTA here -> https://github.com/diyruz/flower/issues/38

shajek, 2023-03-23:
Hello, 

anyone try some of the zigbee 3.0 frimwares from another sensor on diyruz v3 hardware ?