diyruz/flower #43: Request redesign for newer Texas Instruments multi-protocol module like CC2652

Hedda, 2021-10-21:
Have you considered replacing CC2530 with new TI MCU like CC2652 or similar?

Should in theory make the battery last longer as well as make for more future-proof hardware with the potential of perhaps not only using Zigbee 3.0 but also even using Thread, Matter (Project CHIP), and/or Bluetooth via multi-protocol support as well.

Maybe use one of the newer CC2652R or CC2652RB multi-protocol modules from RF-Star or Ebyte?

https://www.szrfstar.com/product/189-en.html

https://www.szrfstar.com/product/list-196-en.html

https://www.ebyte.com/en/product-view-news.html?id=1481

https://www.ebyte.com

bojanpotocnik, 2021-11-29:
This would also make https://github.com/diyruz/flower/issues/20 possible

Hedda, 2022-02-08:
@nurikk Have you given any more thought to @diyruz flower sensor redesign with a CC2652 radio module? 

For your information, there is much discussion about it and its derivative Zigbee flower/plant sensors here:

https://community.home-assistant.io/t/zigbee-flower-sensors-for-soil-moisture-light-temperature-humidity/272803

The "efekta" variants by smartboxchannel has epsecially become popular but it too uses the an old CC2530:

https://github.com/smartboxchannel/Plant-Watering-Sensor-Zigbee

bojanpotocnik, 2022-02-09:
What about EFR32MG2x? I'm lately working with Silabs chips and thinking about doing this with MG2x series...