diyruz/flower #44: Failing to fully pair in zigbee2mqtt

robertalexa, 2021-11-07:
Hi @nurikk 

I have 20 devices based on the V3 specs, built by JLC PCB, all of them running AAA batteries, without the DS or BCM sensors. All devices have been flashed to 2.1.0 version (latest as of now).

I have tried pairing the devices to zigbee2mqtt, and every single time something fails. The devices get paired "successfully", show us supported devices, but the binds fail, or if the binds succeed, the reporting fails, The most I have managed to get them so far was to have battery reporting (though it started with 128%).

I get no values for soil_humidity or illuminance. But like i said, the reporting fails, so I am not surprised.

I have tried pairing the devices in a random location, then close to a router device, then close to the coordinator, same result. After removing the device (force) from z2m, I have always restarted z2m.

So I am lost here. Would you be able to help, or do I basically have 20 useless devices now?

Thanks in advance

mathieuchurlet, 2021-11-07:
I have the same problem. 
Sometimes, the device get paired successfully, and for 5 devices, i have all the values, but for others, interview ok, but no value are sent.

robertalexa, 2021-11-07:
I have just tried to pair one to my testing environment on a CC2531 and the devices paired and configured successfully. At least I know there is no problem with the device.

Here are screenshots of some info;

Binds
[image]

Reporting
[image]

Exposes
[image]

I will continue digging, it might be just poor quality signal, or interference i guess, but that makes me wonder what the reliability of the device is

nordeep, 2021-11-10:
Had the same issue. In my case some of tricky settings of Zigbee2MQTT gave the result:
Settings - Serial - Adapter - force to zstack
Settings - Serial - Adapter - Disable led.

I'm using cheap CC2531 USB Coordinator. Flashed by CC2531_DEFAULT_20201127 firmware.
https://www.zigbee2mqtt.io/guide/adapters/flashing/alternative_flashing_methods.html

robertalexa, 2021-11-10:
I have done some more testing yesterday with my testing environment, usb in laptop, device next to laptop. All 20 devices paired and configured successfully.

So the issue must just be distance, I will try to get them onto my main network and see what i get

mathieuchurlet, 2021-11-11:
Pairing done with the new sonoff dongle, with a cheap cc2530 with the latest firmware and with a diy CC2538... Nothing with the value. 
The pairing is ok, but the device doesn't send anything...

[image]
[image]
[image]

mathieuchurlet, 2021-11-11:
Same issue if I try to pair near the coordinator or the router.

robertalexa, 2021-11-15:
So I have finally found the time to play with them further.

The devices paired and configured successfully to my main network and have connected via a router. So not sure what happened there at first. Maybe just a bad connection or interference

closed 2021-11-15 by robertalexa