diyruz/flower #50: Short battery life with CR2032

duhanebel, 2022-04-08:
Hi, 
I've made 5 of these sensors v2 (great work btw! Thanks!!) without any extra sensor. That means I've only soldered:
- E18-MSI-PCB
- R1, R2, R3, R4
- LED1
- C1, C2
- D1
- SW1

Unfortunately I've now been running them for a few weeks and the battery voltage seems down between 2.2v and 2.8v. This is not consistent between different sensors.
What can be the reason for this? How can I find out what has gone wrong?

quark72, 2022-05-22:
I have the same problem and discovered, that the module send not only one telegram/message. It send between 4 and 6 telegrams. And i see how in the first telegram the Voltage is 3.0V and in the last telegram it drops down. 
I don't understood when and why how much telegrams are repeated - sometime it send 4, sometime 6, sometime 4. Only one time i received only one telegram.
The sending takes sometime a little bit over 1sec. Is it possible that this repeatedly sending of Data the Problem is?

mattes34, 2022-06-04:
I also had the problem and built a firmware where only the soil moisture is measured. With this firmware I don't see jumps in the battery value since one day.

duhanebel, 2022-06-05:
@mattes34 would you be able to send me an hex file with your version? I can't make IAR work for me no matter how much I tried :/

mattes34, 2022-06-05:
Sure [DIYRuZ_Flower_only_soilmoisture.zip](https://github.com/diyruz/flower/files/8840707/DIYRuZ_Flower_only_soilmoisture.zip)
I think only with soil moisture measuring its possible to save energy by going back to only one cycle measurement like it was before [Do only one wakeup cycle](https://github.com/diyruz/flower/commit/340a96fce2826ec9c99b3dd8207342343cedc8ef). If I fine time I will test it in the next days

smartboxchannel, 2022-06-07:
Try this special firmware for Versions 2 and 3 sensors running on cr2032

https://github.com/smartboxchannel/Plant-Watering-Sensor-Zigbee/commit/53922d31bb3e81a2c23cdb097d4571fa9a04cc76

duhanebel, 2022-06-13:
@smartboxchannel is that a drop in replacement for the v2 I've built? I've tried to flash it but the device looks dead afterward

MatthijsThijssen, 2022-06-13:
Dear sender,

I will be out of the office starting 9 June through 27 June returning. During this period i have limited access to my email. I will respond to your emails as soon as possible upon my return.
You can also send your message to ***@***.*** so one of my colleague can follow up.
Thank you for your message

duhanebel, 2022-06-21:
@mattes34 I've tried with your firmware and while one device now works, other two refused to pair (3 LED blink every time). I'm not sure what I'm doing wrong. I've tried resetting the board by holding the button for 10seconds. On zigbee2mqtt I forced-removed the devices. My coordinator has only 10 devices connected so I don't think that is the problem (it wasn't before)

Darozas, 2022-12-12:
> @mattes34 I've tried with your firmware and while one device now works, other two refused to pair (3 LED blink every time). I'm not sure what I'm doing wrong. I've tried resetting the board by holding the button for 10seconds. On zigbee2mqtt I forced-removed the devices. My coordinator has only 10 devices connected so I don't think that is the problem (it wasn't before) **UPDATE:** Never mind - my coordinator wasn't powerful enough and now it all works!

Hi duhanebel , And how is battery life is with this firmware ? Regards

mattes34, 2022-12-12:
Hi, I think the runtime was still not good. All my sensors run out after ~30days (2h reporting time). My new Idea is to set more focus on the measurement itself, becouse the start and stop of sampling is done in the event loop (static void zclApp_ReadSensors(void)) and maybe the measurement PWM is to long on. But I don't find time to work on it. To many projects.

mattes34, 2022-12-12:
Hi, I make some changes (dirty and untested).  good luck :D

https://github.com/mattes34/flower/blob/master/firmwares/DIYRuZ_Flower_soil_moisture_only.hex

Darozas, 2022-12-12:
> Hi, I make some changes (dirty and untested). good luck :D
> 
> https://github.com/mattes34/flower/blob/master/firmwares/DIYRuZ_Flower_soil_moisture_only.hex

Thanks Got it installed ill let you know results in few weeks!!!

duhanebel, 2022-12-13:
I have a strange experience: out of 20 sensors, almost half have been running for 3+ months and have still a good amount of battery. The other half dies within weeks with brand new batteries. I wonder if it might have to do with my soldering? The components are all from the same batch.
I’ll try the new firmware anyway.. worth a try!

mattes34, 2022-12-13:
Same manufacturer of the batteries  ?
@Darozas I will build a version in the next days.

What I forgot to say is that I changed the compensation in this firmware version. :-/ 
I make some measurements and use a curve that 0 to 100%  is from wet soil to dry. Before it was always from air to wet soil

duhanebel, 2022-12-13:
Same battery pack even. I’ve changed the firmware at one point to measure every 15m and on some sensors, the 15m firmware performs better than the 30m ones.. so I’m really unsure what’s happening.
Happy to test the firmware on the “bad ones” and see if it improves.

Darozas, 2022-12-13:
> Same manufacturer of the batteries ? @Darozas I will build a version in the next days.
> 
> What I forgot to say is that I changed the compensation in this firmware version. :-/ I make some measurements and use a curve that 0 to 100% is from wet soil to dry. Before it was always from air to wet soil

Thanks Cant wait to try!!!!

duhanebel, 2023-01-05:
I've just identified one of the "bad sensors". I'll install today with a fresh batter and then we'll see what happens (with the previous firmware that board died in 40 days)

Darozas, 2023-01-05:
I have one sensor running with latest "Untested Firmware" 

and battery didn't dropped yet. Its hard to tell which firmware is on sensor as both firmware's have same data 

If you dont make note you can easily forget which is which

[image]

duhanebel, 2023-01-07:
I'm having a hard time pairing up the sensor after flashing.. my distance of zibgee2mqtt doesn't see it at all.. mmh

duhanebel, 2023-02-27:
I’ve got one device running on this firmware since jan and so far the battery seems to be steady at 3v.
Unfortunately I’m having a hard time flashing more devices: is there a particular car I need to take when reflashing? With the new firmware the device never completely configure with zigbee2mqtt and pressing the button on the pcb does nothing (the led doesn’t flash)

duhanebel, 2023-10-25:
After 6 months of testing I can conclude that... nothing. The behaviour seems to be erratic: some boards last months with one batter and then weeks with another. Some boards go on for months, while other die consistently in a few weeks. I've used the same kind of components for all 20 of them and soldered them all the same day so they should behave pretty similarly to each other.
I think this is a limitation of the CR2032: they're not consistently charged with some batches/brand lasting longer than others. This kind of device requires a battery well charged to operate (> 1.8v remaining - I can happily reuse depleted batteries with wall clocks and stuff and they'd go on for months) and I think that just makes this situation worse.
I might just swap them all for the AAA holder: that would boost capacity 5x-8x and hopefully solve the problem.

At least that's my conclusion after all these months.. let me know if you had a different experience.

AlirezaT, 2025-02-22:
I made 3 devices with all sensors and all three can't last more than a couple of months with the CR2032 Battery.  when I measure the current consumption, it is not more than 5uA DC but I'm suspicious about the 220uF Tantalum caps which are known for high leakage current. I've replaced them by a 10uF ceramic cap which is very good for leakage but its capacity will drop ~50% at half of the rated voltage. However their ESR is much better than the Tantalum caps . I'll report my finding here 🙂

AlirezaT, 2025-08-01:
@duhanebel  I had 3 sensors, i could confirm that after replacing the 220uF tantalum cap with MLCC, the battery lifetime increased. For one sensor the difference was very significant. However, still they don't last for more than 3 to 4 months. 
I'm now thinking of switching the firmware to this: https://modkam.ru/2021/02/21/sensor-vlazhnosti-pochvy-zigbee-versija-3/
It is claimed to be more  efficient !

fabiogallonetto, 2025-11-12:
@AlirezaT How is the switch to v3 going? I hear is complicated to flash. Are you using aaa or cr2032 batteries? What sensors have you soldered on the board?