diyruz/flower #52: Please add v3 into the github

Drizzt321, 2022-09-26:
Found there's a [v3](https://modkam-ru.translate.goog/2021/02/21/sensor-vlazhnosti-pochvy-zigbee-versija-3/?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp) (via Google Translate) in the blog posts, was hoping @nurikk could upload the v3 stuff into here.

Also, for anyone who links out and downloads the v3 from the blog post, JLCPCB, as per [this comment exchange](https://modkam-ru.translate.goog/2021/02/21/sensor-vlazhnosti-pochvy-zigbee-versija-3/?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp#comment-6595), the R2 specified in the BOM, for some reason JLCPCB/LCSC was changed to be size of 0405 with 1K ohm, not a 0805 1M ohm as the BOM indicates. As per commenter Alexander, I'm trying to use component [0805W8F1004T5E ](https://lcsc.com/product-detail/Chip-Resistor-Surface-Mount_UNI-ROYAL-Uniroyal-Elec-0805W8F1004T5E_C17514.html), although haven't gotten the boards yet to verify things are working.

MatthijsThijssen, 2022-09-26:
Dear sender,

I will be out of the office starting 22 Sept through 27 Sept returning. During this period i have limited access to my email. I will respond to your emails as soon as possible upon my return.
You can also send your message to ***@***.*** so one of my colleague can follow up.
Thank you for your message

BenRoe, 2023-12-22:
Any update on this and what are the changes from v2 to v3?