diyruz/flower #54: BME280 sensor not reporting data

Darozas, 2022-11-25:
Hi All 

I have sensor mounted directly on board but get no readings

[image]

and it looks like some components missing on pcb for sensors?

[image]

recommended 

DS sensor should have 4.7K resistor but its not on beard 

same with BME280 no 2 resistors missing 

[image]

Is that just me or some one else have same issue here with sensors?

Regards

nocodeman55, 2023-03-24:
Did you ever solve this? I also have a BME280 SMD directly to the board and it gives no readings

machinev1, 2024-09-14:
I was confronted with this after the first one was made. Does anyone have any new info, solutions or anything that might help since then? I have 20 PCBs ready at home and I don't want them all to go in the trash...